Nitcloud/Digital-IDE #1: Cannot startfpga

Pro-A, 2020-03-09:
i meet some problem like this
[https://forums.xilinx.com/t5/Vivado-TCL-Community/Using-Python-within-TCL-script-in-Vivado-2019-1/m-p/982047#M8125](Using Python within TCL script in Vivado 2019.1)

I wonder if I need to use python2?
(But when I tried to use python 2.7.5, I met some other problem)

I will add screenshot and log later.

Pro-A, 2020-03-10:
Well, I solved the python problem by add the following lines in tcl script.

```tcl
unset ::env(PYTHONPATH)
unset ::env(PYTHONHOME)

set ::env(PYTHONPATH) "C:/Program Files/Python38/python38.zip:C:/Program Files/Python38/DLLs:C:/Program Files/Python38/lib:C:/Program Files/Python38:C:/Program Files/Python38/lib/site-packages"
set ::env(PYTHONHOME) "C:/Program Files/Python38"
```
But there are still some other problems.

[image]

I think maybe there are something wrong when initializing the project.

Pro-A, 2020-03-10:
After comment-out the codes, the script can run now.

[image]

[image]

And I have to recover it after adding IP cores, I think.

Bestduan, 2020-03-10:
Thank you for your question. This plug-in was not designed for IP and bd at the beginning, while the IP of cortexM3 was added temporarily. I also encountered some problems when I used it today, and I will fix the bug in the near future.In addition, I used version 3.6 of python, with the environment variable added.Finally, if you have any questions, please feel free to ask here.

Bestduan, 2020-03-10:
In addition, I recommend that you can change the “glob” directive to the “glob-nocomplain /user/Hardware/ src/*.bd”. 
Like:
foreach bd_file [glob -nocomplain ./user/src/*.bd] {
		add_file $bd_file -quiet
}
Because the direct add_file directive cannot be added to the project when bd files are in the src folder.

Bestduan, 2020-03-10:
The bd design was removed in the updated version 0.1.4.Also, if the python version is 3.6 and the environment variable is not available after it has been set, remove comments about python environment variables from the plug-in's internal scripts
[image]

closed 2020-04-25 by Pro-A