Nitcloud/Digital-IDE #5: Succeed using module pyperclip to control clipboard

Pro-A, 2020-04-27:
## Description
I tried using pyperclip to copy instance to clipboard, and succeeded in macOS.

I am not sure if it makes sense on other os. I will try it later.

## Environment
- OS: macOS 10.15.4
- Python: 3.7.7

## Solution
Use a string to record data, for example, ```instance_data```.
And then just use
```python
pyperclip.copy(instance_data)
```
to copy the string to clipboard.

Bestduan, 2020-04-27:
I simply tested that this function is valid in win, but it can only copy the part of the example, the part of the port is not copied in.
[image]
[image]
[image]

Pro-A, 2020-04-27:
> I simply tested that this function is valid in win, but it can only copy the part of the example, the part of the port is not copied in.
> [image]
> [image]
> [image]

I use one string to store the whole data.
Maybe you can also try this.
```python
def writeTestBench(input_file):
    """ write testbench to file """
    with open(input_file, 'rb') as f:
        f_info = chardet.detect(f.read())
        f_encoding = f_info['encoding']
    with open(input_file, encoding=f_encoding) as inFile:
        inText = inFile.read()

    # removed comment,task,function
    inText = delComment(inText)
    inText = delBlock(inText)

    # moduel ... endmodule  #
    moPos_begin = re.search(r'(\b|^)module\b', inText).end()
    moPos_end = re.search(r'\bendmodule\b', inText).start()
    inText = inText[moPos_begin:moPos_end]

    name = findName(inText)
    paraList = paraDeclare(inText, 'parameter')
    paraDec, paraDef = formatPara(paraList)

    ioPadAttr = ['input', 'output', 'inout']
    input = portDeclare(inText, ioPadAttr[0])
    output = portDeclare(inText, ioPadAttr[1])
    inout = portDeclare(inText, ioPadAttr[2])

    portList = formatPort([input, output, inout])
    input = formatDeclare(input, 'reg')
    output = formatDeclare(output, 'wire')
    inout = formatDeclare(inout, 'wire')

    # write Instance
    instance_data = ""
    # module_parameter_port_list
    if (paraDec != ''):
        print("// %s Parameters\n%s\n" % (name, paraDec))
        instance_data += "// " + name + " Parameters\n" + paraDec + "\n"

    # list_of_port_declarations
    print("// %s Inputs\n%s\n" % (name, input))
    instance_data += "// " + name + " Inputs\n" + input + "\n"
    print("// %s Outputs\n%s\n" % (name, output))
    instance_data += "// " + name + " Outputs\n" + output + "\n"
    if (inout != ''):
        print("// %s Bidirs\n%s\n" % (name, inout))
        instance_data += "// " + name + " Bidirs\n" + inout + "\n"

    print("%s %s U_%s (\n%s\n);" % (name, paraDef, name, portList))
    instance_data += name + " " + paraDef + " U_" + name + "_0 (\n" + portList + "\n);"
    pyperclip.copy(instance_data)
```

closed 2020-05-19 by proarz