Nitcloud/Digital-IDE #6: Code refactoring

amitkulkarnis, 2020-04-27:
Is it possible to refactor a signal/port name present in multiple modules (files: *.v)?

Bestduan, 2020-04-28:
I don't quite understand what you are saying, I hope you can give an example if possible. In addition, I think this is related to the code syntax, not to the plug-in.

amitkulkarnis, 2020-04-28:
E.g. if I want to change the name of a signal or port that is used in multiple HDL files then the refactor should take care of replacing the new name in corresponding places of all the modules.

Bestduan, 2020-04-28:
I see what you mean. That's a good idea, but it's still hard for me to implement. I will consider adding this feature later.

StrideZhou, 2023-03-06:
@amitkulkarnis  you can use the "change all current" to do that (provide by VSC)