Nitcloud/Digital-IDE #7: HoverProvider.js error.

crs-ica, 2020-05-15:
Hello,

First of all, congrats for your work. I think you had a very good idea to combine FPGA development languages support in one extension.

I am running VS Code 1.45.1.
When hoovering the code (.sv, .v), the following error is printed to console:
 
[2020-05-15 11:59:28.926] [renderer1] [error] An unknown error occurred. Please consult the log for more details.
[2020-05-15 11:59:28.927] [renderer1] [error] Cannot read property 'uri' of undefined: TypeError: Cannot read property 'uri' of undefined at c:\Users\userx\.vscode\extensions\sterben.fpga-support-0.1.10\src\.Providers\providers\HoverProvider.js:13:67 	at processTicksAndRejections (internal/process/task_queues.js:85:5)

Bestduan, 2020-05-15:
[image]
As shown in the screenshot, there was no error in my hover prompt.And I also used 1.45.1, however the system is win10.If possible, I hope you can provide error reporting screenshots so as to better solve the problem.

crs-ica, 2020-05-15:
At program startup:

[image]

At mouse hoover:
[image]

I also have win10.
If you need more info, I'll be glad to assist.

Bestduan, 2020-05-16:
As I know it's maybe a bug in VSCode.I'm sorry that I can't solve your problem. I've tried many cases without your problem.My vivado version is 18.3, but I don't think this is a problem with the vivado version.In addition, this part of the functionality is directly integrated with the **SystemVerilog - Language Support** plug-in, you can try to install this plug-in to see if the same problem occurs, if not, you can try to use the two plug-ins together.Finally try lowering the vscode version and try again.

crs-ica, 2020-05-16:
I have also Eirik's extension and I saw that I have the same error with it.

Bestduan, 2021-04-18:
This issue has been addressed in the new version，you can check it.

closed 2021-04-18 by Bestduan