Nitcloud/Digital-IDE #8: TCL syntax highlighting: 'foreach'

Nabav, 2020-06-03:
The `foreach` [command](https://www.tcl.tk/man/tcl8.6/TclCmd/foreach.htm) is not defined in TCL syntax definition [file](https://github.com/Bestduan/fpga_support_plug/blob/master/syntaxes/tcl.json). Therefore, the `for` part is highlighted and the remaining `each` part is left unattended.

Bestduan, 2020-06-03:
{
  "name": "tcl",
  "scopeName": "source.tcl",
  "fileTypes": [
    "tcl",
    "xdc",
	"sdc",
	"fdc"
  ],
  "uuid": "45aff42e-4fcd-4717-829b-4cdaf9d707aa",
  "variables": null,
  "patterns": [
	{
		"captures": {
		  "1": {
			"name": "keyword.control.tcl"
		  }
		},
		"match": "(?<=^|[\\[{;])\\s*(if|while|for|catch|default|return|break|continue|switch|exit|foreach|try|throw)\\b"
	},
    {
      "include": "#comment"
    },
    {
      "include": "#command"
    }
  ],
  "repository": {
    "comment": {
      "patterns": [
        {
          "name": "comment.line.tcl",
          "match": "\\s*\\#.+"
        }
      ]
    },
    "command": {
      "patterns": [
        {
          "include": "#tcl_commands"
        },
        {
          "include": "#default_command"
        }
      ]
    },
    "expression": {
      "patterns": [
        {
          "include": "#variable"
        },
        {
          "include": "#brackets"
        },
        {
          "name": "constant.numeric.tcl",
          "match": "(?:0(?:x|o|b))?\\d+(?!\\.)"
        },
        {
          "name": "constant.numeric.tcl",
          "match": "\\d+\\.(?:\\d+(?:[eE][+-]?\\d+)?)?"
        },
        {
          "name": "keyword.operator.tcl",
          "match": "(?:\\+|\\-|\\~|\\!|\\*\\*|\\*|\\%|\\<\\<|\\>\\>|\\<\\=|\\>\\=|\\<|\\>|\\=\\=|\\!\\=|eq|ne|in|ni|\\&\\&|\\&|\\|\\||\\||\\^|\\?|\\:)"
        }
      ]
    },
    "double_quotes": {
      "patterns": [
[... 316 lines cut ...]
          ]
        }
      ]
    },
    "tcl_commands": {
      "patterns": [
        {
          "include": "#proc"
        },
        {
          "include": "#for"
		},
		{
		  "include": "#foreach"
		},
        {
          "include": "#if"
        },
        {
          "include": "#expr"
        }
      ]
    }
  }
}

Bestduan, 2020-06-03:
You can try changing the Tcl.json file to the above configuration

Nabav, 2020-06-03:
> You can try changing the Tcl.json file to the above configuration

I have fixed the escape characters and tested. It is not working again, since the `"foreach":` entry is missing under `"repository":`. I do not know how to define the pattern characters for a new command.

Nabav, 2020-06-03:
@Bestduan, we need something like this, but it is nor correct. Can you fix please?
```json
    "foreach": {
      "patterns": [
        {
          "name": "meta.command.foreach.tcl",
          "begin": "(foreach)",
          "beginCaptures": {
            "1": {
              "name": "keyword.control.foreach.tcl"
            }
          },
          "end": "(?:(;)|(?=\\})|(?=\\])|$)",
          "endCaptures": {
            "1": {
              "name": "punctuation.end.semicolon.tcl"
            }
          },
          "patterns": [
            {
              "begin": "(?=\\{)(?=(?:.+?\\{))",
              "end": "(?<=\\})",
              "patterns": [
                {
                  "include": "#expr_braces"
                }
              ]
            },
            {
              "begin": "(?=\\{)",
              "end": "(?<=\\})",
              "patterns": [
                {
                  "include": "#tcl_braces"
                }
              ]
            },
            {
              "include": "#brackets"
            },
            {
              "include": "#braces"
            }
          ]
        }
      ]
    },
```

Bestduan, 2020-06-03:
I have tried to fix it in the way you mentioned, but again it doesn't work, so I have added the following code to patterns so that we can highlight foreach. I am not familiar with the highlight grammar itself, and I just refer to the relevant design for reference.
{
	"captures": {
	    "1": {
		"name": "keyword.control.tcl"
	    }
	},
	"match": "(?<=^|[\\[{;])\\s*(if|while|for|catch|default|return|break|continue|switch|exit|foreach|try|throw)\\b"
},

Nabav, 2020-06-03:
Then we can probably wait for a feedback from:
https://github.com/go2sh/tcl-language-support/issues/2

Thanks for the support. :heart:

go2sh, 2020-07-03:
Hi, i fixed some of the highlighting especially foreach while for if/else break continue and some dict functions. Next step is to add try/trap/finally switch exit throw and lmap.

Bestduan, 2021-04-18:
This issue has been addressed in the new version

closed 2021-04-18 by Bestduan