Nitcloud/Digital-IDE #11: 右键点击start fpga么有反应

qgzln, 2021-02-26:
如题，但是可以在命令行里用Python操作，下载插件地址在C盘的默认的.vscod里.vscode\extensions\sterben.fpga-support-0.1.10\src\.TOOL\.Script\start.py

Bestduan, 2021-02-26:
start.py的实际路径应该是.vscode\extensions\sterben.fpga-support-0.1.10\src\.TOOL\.Script\start.py  插件也是直接从该路径下启动start.py，如果路径错误就会启动不了，是否可以将.vscode\extensions\sterben.fpga-support-0.1.10\src下的内容截图从而更好地解决问题

Bestduan, 2021-02-26:
start.py的实际路径应该是.vscode\extensions\sterben.fpga-support-0.1.10\src\\.TOOL\\.Script\start.py 插件也是直接从该路径下启动start.py，如果路径错误就会启动不了，是否可以将.vscode\extensions\sterben.fpga-support-0.1.10\src下的内容截图从而更好地解决问题

Bestduan, 2021-02-27:
或者使用快捷键的方式Alt+Z启动，是否有用

qgzln, 2021-02-27:
> 或者使用快捷键的方式Alt+Z启动，是否有用

[image]
用快捷键也不行说找不到

qgzln, 2021-02-27:
> start.py的实际路径应该是.vscode\extensions\sterben.fpga-support-0.1.10\src\.TOOL\.Script\start.py 插件也是直接从该路径下启动start.py，如果路径错误就会启动不了，是否可以将.vscode\extensions\sterben.fpga-support-0.1.10\src下的内容截图从而更好地解决问题

路径是这个路径，但是vscode里启动不了，必须手动，而且在ctrl+shift+p里面没看到这个命令

Bestduan, 2021-02-27:
你可以尝试在.vscode\extensions\sterben.fpga-support-0.1.10\package.json文件里查找是否有"command": "extension.StartFPGA",
[image]
以及查看.vscode\extensions\sterben.fpga-support-0.1.10\src\extension.js中的注册代码是否正确，如下图
[image]
与startfpga相关的基本就这两个地方
如果没有可以尝试重装插件

qgzln, 2021-02-28:
> 你可以尝试在.vscode\extensions\sterben.fpga-support-0.1.10\package.json文件里查找是否有"command": "extension.StartFPGA",
> [image]
> 以及查看.vscode\extensions\sterben.fpga-support-0.1.10\src\extension.js中的注册代码是否正确，如下图
> [image]
> 与startfpga相关的基本就这两个地方
> 如果没有可以尝试重装插件

这两个地方没有问题，重新装了还是老问题，右键没反应，快捷键报错，右键里面像testbench的另外几个命令也没有反应

Bestduan, 2021-02-28:
那大概率是vscode本身的问题了，我最近插件开发的时候经常遇到奇奇怪怪的问题，感觉这几版的vscode不是很稳定，你尝试回退vscode版本试试，实在不行我也没办法了

qgzln, 2021-03-01:
> 那大概率是vscode本身的问题了，我最近插件开发的时候经常遇到奇奇怪怪的问题，感觉这几版的vscode不是很稳定，你尝试回退vscode版本试试，实在不行我也没办法了

好的，那就等下个版本vscode吧

qgzln, 2021-03-01:
> 那大概率是vscode本身的问题了，我最近插件开发的时候经常遇到奇奇怪怪的问题，感觉这几版的vscode不是很稳定，你尝试回退vscode版本试试，实在不行我也没办法了

会不会和python有关系，我是用anaconda安装的python

Bestduan, 2021-04-17:
近期调试插件的时候遇到了这个问题，导致的原因是插件没有启动激活，我记得是要打开Verilog文件后才会激活插件

Bestduan, 2021-04-17:
如果想要打开vscode就激活插件更改.vscode\extensions\sterben.fpga-support-0.1.10\package.json里的activationEvents为
"activationEvents": [
		"*"
	],
即可

Bestduan, 2021-05-07:
之前那个问题是说明在workspace下文件结构解析不到，导致插件启动失败，现在已经发布新版本，请问是否还有相同问题？

Bestduan, 2021-05-08:
什么样的状态图显示，有效果图吗

qgzln, 2021-05-08:
应该是状态机图…………TerosHDL这个插件有很多有意思的功能，可以参考下^-^

Bestduan, 2021-05-08:
TerosHDL的转态机图只支持vhdl不支持Verilog，之后有时间会实现一下，现在没时间了

qgzln, 2021-05-08:
> TerosHDL的转态机图只支持vhdl不支持Verilog，之后有时间会实现一下，现在没时间了

已经支持Verilog了，只是生成代码好像只支持vhdl

Bestduan, 2021-05-08:
另一方面他的解析和我的解析不一样，他是调包我是自己写的正则表达做的解析要移植的话工作量不小，以后有时间再说吧，而且TerosHDL装一次就400M，有点难受。我也不是专门写ts和js的，做到现在本来就是兴趣，过程中也遇到了设计瓶颈，现在没啥反馈，感觉没啥动力了，准备佛系更新了，半年后看能不能上岸，上岸再新增feature了，如果喜欢的话麻烦推荐一下，感谢

qgzln, 2021-05-08:
已经推荐刚来的同事们转到vscode下，并且安装了你开发的这个插件了。他那个确实大，而且有时候会出现开发人员工具重启等一系列的bug，现在只有看状态机图的时候用到他这个插件了

qgzln, 2021-05-08:
还有祝福你成功上岸

Bestduan, 2021-05-08:
谢谢，之后我将关闭这个comment，如果贵司在插件使用过程中遇到其他问题可以邮件联系我。

Bestduan, 2021-05-08:
你好，考虑到贵司生产方便，我花了点时间做了一下简单移植（仅状态机显示与设计），包大小为10M，安装后45M左右，暂不平台发表，若有兴趣尝鲜，可以留下你的联系方式，私发给你

closed 2021-05-09 by Bestduan