Nitcloud/Digital-IDE #12: Extension issue

Gold1169, 2021-04-29:
- Issue Type: `Bug`
- Extension Name: `fpga-support`
- Extension Version: `0.1.13`
- OS Version: `Windows_NT x64 10.0.19042`
- VS Code version: `1.55.2`

:warning: We have written the needed data into your clipboard. Please paste! :warning:

Gold1169, 2021-04-29:
```json 
{
	"messages": [],
	"activationTimes": {
		"codeLoadingTime": 9,
		"activateCallTime": 1337,
		"activateResolvedTime": 0,
		"activationReason": {
			"startup": true,
			"extensionId": {
				"value": "sterben.fpga-support",
				"_lower": "sterben.fpga-support"
			},
			"activationEvent": "*"
		}
	},
	"runtimeErrors": [
		{
			"name": "Error",
			"message": "ENOENT: no such file or directory, stat 'c:/NetFirm4.0_svn/src/logic/src/DfaMatch//sim/work/_lock'"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "Error",
			"message": "ENOENT: no such file or directory, stat 'c:/NetFirm4.0_svn/src/logic/src/DfaMatch//sim/work/_lock'"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
[... 313 lines cut ...]
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "TypeError",
			"message": "t is not iterable"
		},
		{
			"name": "Error",
			"message": "ENOENT: no such file or directory, stat 'c:/NetFirm4.0_svn/src/logic/src/DfaMatch//sim/work/_lock'"
		},
		{
			"name": "Error",
			"message": "ENOENT: no such file or directory, stat 'c:/NetFirm4.0_svn/src/logic/src/DfaMatch//sim/work/_lib1_7.qpg'"
		}
	]
}
```

Bestduan, 2021-04-29:
Please describe in detail which feature is not available. Pasting content alone cannot locate the identified bug point

Bestduan, 2021-07-28:
I have released the latest version, in the new version, I focused on optimizing the kernel, whether does it solve your problem?

Bestduan, 2021-07-28:
If you do not respond within a week I will close the issue

closed 2021-08-05 by Bestduan