Nitcloud/Digital-IDE #13: High CPU usage

nchokoev, 2021-05-07:
Issue Type: <b>Bug</b>

This extension causes high CPU utilization.

Extension version: 0.1.15
VS Code version: Code 1.55.2 (3c4e3df9e89829dce27b7b5c24508306b151f30d, 2021-04-13T09:35:57.887Z)
OS version: Windows_NT x64 10.0.19042

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz (12 x 2208)|
|GPU Status|2d_canvas: enabled<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>oop_rasterization: enabled<br>opengl: enabled_on<br>protected_video_decode: enabled<br>rasterization: enabled<br>skia_renderer: enabled_on<br>video_decode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled|
|Load (avg)|undefined|
|Memory (System)|31.73GB (16.41GB free)|
|Process Argv|--crash-reporter-id 23accab7-e812-4840-9c04-35e9f4879f8d|
|Screen Reader|no|
|VM|13%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vsreu685:30147344
python383:30185418
pythonvspyt700cf:30270857
pythonvspyt602:30294772
vspor879:30202332
vspor708:30202333
vspor363:30204092
vstes627:30244334
pythonvspyt639:30291489
pythontb:30283811
vspre833cf:30267465
pythonptprofiler:30281270
vshan820:30294714
pythondataviewer:30285071
vscus158:30286553
vscgsv2ct:30294353

```

</details>

Bestduan, 2021-05-07:
CPU utilization is related to the number of HDL files under your project, and if there are no properties files, all HDL files in this workspace will be directly parsed, and the more files there are, the more computing power will be needed to parse.We will do optimization in the future.

klcheungaj, 2021-06-24:
In my case, the CPU usage became high after some output files were generated. This seems that not only HDL files(.v, .sv, .vhd, etc) but also other text files are parsed.

Bestduan, 2021-07-28:
I have released the latest version, in the new version, I focused on optimizing the kernel,  whether does it solve your problem?

Bestduan, 2021-07-31:
I will close the issue after three days and consider your problem solved

closed 2021-08-05 by Bestduan