Nitcloud/Digital-IDE #14: vscode上使用git插件，会一直显示hdl indexing

qgzln, 2021-05-18:
[image]
如图

Bestduan, 2021-05-18:
你的意思是把git插件卸载后就没有这个问题了？这个应该是和你工程下HDL文件数量相关的，你现在工程里有多少HDL文件？文件结构是怎样的

qgzln, 2021-05-18:
[image]类似这种情况，git插件是一直在的，如果我初始化git库就没这个问题。这个图显示即使我这个工程里只有python的git，也会出现这种情况导致git的插件里的文件刷不出来

qgzln, 2021-05-18:
说错了，是如果不初始化git的存储库就没这个问题

Bestduan, 2021-05-18:
你现在工程里有多少HDL文件？

qgzln, 2021-05-18:
40多个

qgzln, 2021-05-18:
我打开python的工程，也会出现这个情况，只有我不使用git，才正常

Bestduan, 2021-05-18:
请尝试vscode回溯到1.54版本试试，看是否依然出现这个问题

Bestduan, 2021-05-18:
是今天才出现的情况吗，我发现又有一个工程之前一直不会出现这情况，今天突然就出现了

Bestduan, 2021-05-18:
问题找到了，就是HDL文件太多了，就拿cocotb举例，里面有72个HDL文件，又由于是全解析，所以就很解析很长时间，本身又是正则表达式，目前只能用在property文件里声明"HardwareSrc": "./bin",指向需要解析的文件从而不进行HDL的解析，至于为什么不初始化git存储库就没这个问题就想不通了

Bestduan, 2021-05-18:
vscode每次更新都会出很多问题

qgzln, 2021-05-18:
上午回完你，就外出了，上午打开的cocotb到现在还在HDL indexing……这个现象出现了一段时间了，一直没来的及跟你反馈，好像只要git开启了才会有这个问题，如果像下图一样就不会出现一直HDL indexing不停的情况
[image]

qgzln, 2021-05-18:
property文件在哪里打开？

Bestduan, 2021-05-18:
cocotb我是clone下来就有这个问题，property.json是自己生成的

[image]

Bestduan, 2021-05-18:
建议先版本回退看看

qgzln, 2021-05-18:
好的

Bestduan, 2021-05-18:
请问是否已经解决该问题，如果未解决请等待一下版本，我将用webassembly重构整个工程提升速度和效率以支持多文件大工程

qgzln, 2021-05-19:
等下个版本

Bestduan, 2021-07-28:
我已经发布新的版本， 在新版本中着重优化了插件内核，解决了一些可能导致卡死的原因，并且新增了一些功能，不知道是否能够解决你的问题。关于状态机的显示将在下个版本，即明年一月发布，届时将会准备全面替代并超越terosHDL。

Bestduan, 2021-07-31:
三天后我将关闭该issue，届时则认为已经解决你的问题

qgzln, 2021-07-31:
> 我已经发布新的版本， 在新版本中着重优化了插件内核，解决了一些可能导致卡死的原因，并且新增了一些功能，不知道是否能够解决你的问题。关于状态机的显示将在下个版本，即明年一月发布，届时将会准备全面替代并超越terosHDL。

搞定了，期待

closed 2021-07-31 by Bestduan