Nitcloud/Digital-IDE #15: Extension causes high cpu load

Winters123, 2021-05-27:
- Issue Type: `Performance`
- Extension Name: `fpga-support`
- Extension Version: `0.1.15`
- OS Version: `Windows_NT x64 10.0.19042`
- VS Code version: `1.56.2`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`c:\Users\ADMINI~1\AppData\Local\Temp\sterben.fpga-support-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

Bestduan, 2021-05-27:
Thanks for your advice, I have found out the reasons for the high CPU consumption. There are mainly two aspects, one is the HDL syntax check, which causes CPU consumption under multiple files, the other is the execution of the exec function of the regular expression in NodeJS is stuck, so I have to admit that the performance of the regular expression in JS is not very ideal.This will be fixed in a later release.

Winters123, 2021-05-27:
I see. Thanks for your reply.

closed 2021-05-27 by Winters123