Nitcloud/Digital-IDE #16: 无法simulate

mrxzeng, 2021-06-12:
安装本插件后无法在vscode中simulate，已安装iverilog且path环境变量已设置，可于powershell中使用iverilog -o指令编译成功。
但使用本插件在vscode的.v文件界面右键选择Simulate编译时vscode提示如下错误：
'C:/iverilog/biniverilog.exe' �����ڲ����ⲿ���Ҳ���ǿ����еĳ��� ���������ļ���
来源：FPGA Develop Support（扩展）
注：错误信息实际显示即为乱码

其中，vscode的seetings.json相关设置如下：
    "HDL.linting.linter": "iverilog",
    "verilog.runInTerminal": true,
    "HDL.linting.iverilog.arguments": "-i",
    "TOOL.iVerilog.install.path": "C:/iverilog/bin",
    "TOOL.gtkwave.install.path": "C:/iverilog/gtkwave/bin",

相应的，系统变量Path如下：
C:\iverilog\gtkwave\bin
C:\iverilog\bin\

我安装的iverilog链接为“http://bleyer.org/icarus/iverilog-v11-20210204-x64_setup.exe”
操作系统为Win10专业版-20H2,操作系统内部版本	19042.1052,处理器为amd 3700x,

所Simulate的文件内容为下述内容，文件编码为UTF-8.

module v1 (
    input clock
);
    always @(clock) begin
        $display("Hello World!");
    end

endmodule

mrxzeng, 2021-06-12:
注：多文件项目中，可观察到该插件已经分析出了项目中模块间的调用关系

Bestduan, 2021-06-12:
错误日志显示乱码，建议截图显示

mrxzeng, 2021-06-13:
[image]
[image]
[image]
[image]
图示内容为Path设置、settings.json设置、以及插件已成功识别多文件目录结构相关图片

Bestduan, 2021-06-13:
首先加入环境变量之后就不用再在vscode里面设置安装路径，另外安装路径设置的是错的，看报错信息，iverilog的安装路径应该设置为"TOOL.iVerilog.install.path": "C:/iverilog/bin/" gtkwave需要手动启动，插件不会自动打开

mrxzeng, 2021-06-13:
问题解决了，谢谢

closed 2021-06-13 by Bestduan