Nitcloud/Digital-IDE #18: Xilinx environment variable is failed to configure

Tomolin, 2021-07-05:
### Platform
Ubuntu20.04.1 LTS X86_64, 5.8.0-59-generic kernel version
Vivado 2020.2, Vitis_HLS 2020.2
`Python 3.8.10` runs in `GCC 9.4.0`
### Description
Sytax highlight works fine, but no syntax checking or other features from Xilinx development tool after I set up xilinx install path. It is not fixed even if I changed the software version to 2020.3. Neither vivado or xsct command is valid. I found no answers in closed or opened issue, any suggestion is helpful, thanks!

```
settings.json
{
    "terminal.integrated.tabs.enabled": true,
    "explorer.confirmDelete": false,
    "TOOL.xilinx.install.path": "/opt/Xilinx/Vivado/2020.2/bin, /opt/Xilinx/Vitis_HLS/2020.2/bin, /opt/Xilinx/Vitis/2020.3/bin"
}
```

Bestduan, 2021-07-06:
In fact, Tool.xilinx.install. path is not used in the plugin because it needs to be combined with other functionality. You need to add vivado and XSCT to the environment variable

Tomolin, 2021-07-07:
I will stick to working projects on XilinxGUI, but I want to edit source files in VSCode. Now I added permanent environment variables, bash shell detected `vivado` and `xsct`. Still, syntax checking or grammar diagnosis is not working. I changed HDL.LINTING.LINTER settings as xvhdl, and nothing happened on .vhdl files.
### Errors
While I followed the start gif the file **property.json** is not generated properly, the opened folder did not creat any files or directory. And it prints this: `property file already exists!!!`

According to the printing of extension launch, I am not sure if it is a terminal incompatible problem.
```
$ vivado -mode tcl -s /home/tiger/.vscode/extensions/sterben.fpga-support-0.1.15/out/../resources/tool/Xilinx/Script/Vivado/Run.tcl -notrace -nolog -nojournal

****** Vivado v2020.2.2 (64-bit)
  **** SW Build 3118627 on Tue Feb  9 05:13:49 MST 2021
  **** IP Build 3115676 on Tue Feb  9 10:48:11 MST 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

source /home/tiger/.vscode/extensions/sterben.fpga-support-0.1.15/out/../resources/tool/Xilinx/Script/Vivado/Run.tcl -notrace
WARNING: [Vivado 12-818] No files matched '*'
No entry for terminal type "xterm-256color";
using dumb terminal settings.
No entry for terminal type "xterm-256color";
using dumb terminal settings.
```
```
$ xsct
rlwrap: warning: your $TERM is 'xterm-256color' but rlwrap couldn't find it in the terminfo database. Expect some problems.
                                                                                
****** Xilinx Software Commandline Tool (XSCT) v2020.3
  **** SW Build 3173277 on Wed Apr  7 05:07:21 MDT 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.
```

Bestduan, 2021-07-07:
Please look at the location where your property.json was generated. If it is in the .vscode folder, the file is hidden. Also, enter `xvlog --version` and `xvhdl --version` in bash to see if the version information is returned properly. Finally, enter the `vivado -mode gui` to see if the Vivado GUI opens properly.

Bestduan, 2021-07-07:
Please enter the `vivado -mode tcl` command into bash again to see if the Vivado terminal will be forced to exit

Tomolin, 2021-07-07:
property.json is in ~/.vscode, it seems to relate to project set up, which is not my final goals. Though I made a copy of it in the working folder, nothing happened.
```
$ xvlog --version
Vivado Simulator v2020.2.2
$ xvhdl --version
Vivado Simulator v2020.2.2
$ vivado -mode gui

****** Vivado v2020.2.2 (64-bit)
  **** SW Build 3118627 on Tue Feb  9 05:13:49 MST 2021
  **** IP Build 3115676 on Tue Feb  9 10:48:11 MST 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

start_gui
INFO: [Common 17-206] Exiting Vivado at Tue Jul  6 20:28:50 2021...
$ vivado -mode tcl

****** Vivado v2020.2.2 (64-bit)
  **** SW Build 3118627 on Tue Feb  9 05:13:49 MST 2021
  **** IP Build 3115676 on Tue Feb  9 10:48:11 MST 2021
    ** Copyright 1986-2021 Xilinx, Inc. All Rights Reserved.

No entry for terminal type "xterm-256color";
using dumb terminal settings.
No entry for terminal type "xterm-256color";
using dumb terminal settings.
Vivado% 
```
While I click on FPGA tools> FPGA options> GUI, the console in vivado2020.2 prints below.
```
start_gui
source /home/tiger/.vscode/extensions/sterben.fpga-support-0.1.15/out/../resources/tool/Xilinx/Script/Vivado/Run.tcl -notrace
INFO: [IP_Flow 19-234] Refreshing IP repositories
INFO: [IP_Flow 19-1704] No user IP repositories specified
INFO: [IP_Flow 19-2313] Loaded Vivado IP repository '/opt/Xilinx/Vivado/2020.2/data/ip'.
WARNING: [Vivado 12-818] No files matched '*'
```

Bestduan, 2021-07-07:
When you click on FPGA tools> FPGA options> GUI ,does the vivado GUI launch successfully?  So far, your configuration seems to be fine. How did you test the Grammar Diagnosis function? Did you write out an error grammar , save it, and never respond all the time?

Tomolin, 2021-07-07:
1. vivadoGUI launched successfully, just opened the folder without project, no project summary
2. I found it works if I operate on source files that are copied out of project directory, save it and red mark prints.

If I do it in my project folder, it won't work. Keywords is marked in red if I comment process or component declaration. Nothing will happen if I comment signal or sequencial sentenses. I made copy of property.json here, but the situation is different from above, very strange.
However, some other problems still remain unsolved. 1. property.json file generation 2. warning about $XTERM, it seems you are windows developer, and I am a Linux user.
Besides, I looked into property.json file in ~/.vscode, chip selection is very limited, my chip is XC7A100TCSG324-1, artya7-100t board.

Bestduan, 2021-07-07:
The property.json chip support can be added by yourself, or you can input your model number directly. This will warn you but will not prevent the project from being built.In addition, it is recommended for Linux users to place property.json in the root directory of the workspace, which is still valid and will not be hidden.

Bestduan, 2021-07-07:
Finally, the property.json needs to be saved before the project structure can be built.For your chip type property.json example:
```json
{
	"FPGA_VERSION": "xilinx",
	"PRJ_NAME": {
		"FPGA": "tempt"
	},
	"SOC_MODE": {
		"soc": "none"
	},
	"enableShowlog": true,
	"Device": "xc7a100tcsg324-1"
}
```

[image]

You can also use this method to add your chip type to support and restart VS Code to take effect.

Bestduan, 2021-07-07:
For this plug-in, I basically use JS language for refactoring, trying to smooth the gap between the platforms, but unfortunately I do not have Linux environment. I am alone. Thank you for your support and understanding.

Tomolin, 2021-07-08:
I really appreciate your effort on this project, this is the only extension I found that supports xilinx development tools in vscode market. I searched around and I think windows developer are majority user of xilinx tools, little thread or post talks about software problems in linux platform. 
However, I finally managed to made vscode as vivado text editor, and it works stable now. I configured the settings within vivado, choosed custom editor, filled in `code [file name]`, then everything goes well. I might explore further with this extension, for now I don't need to migrate project environment into vscode. And thank you for your work, this project should take time for good doc and powerful features.

closed 2021-07-08 by Tomolin