Nitcloud/Digital-IDE #19: Extension causes high cpu load

GamearooCodes, 2021-07-18:
- Issue Type: `Performance`
- Extension Name: `fpga-support`
- Extension Version: `0.1.15`
- OS Version: `Windows_NT x64 10.0.22000`
- VS Code version: `1.59.0-insider`

:warning: Make sure to **attach** this file from your *home*-directory:
:warning:`c:\Users\Gamea\AppData\Local\Temp\sterben.fpga-support-unresponsive.cpuprofile.txt`

Find more details here: https://github.com/microsoft/vscode/wiki/Explain-extension-causes-high-cpu-load

GamearooCodes, 2021-07-18:
[sterben.fpga-support-unresponsive.cpuprofile.txt](https://github.com/Bestduan/fpga_support_plug/files/6837119/sterben.fpga-support-unresponsive.cpuprofile.txt)

Bestduan, 2021-07-19:
Check to see if the issue has occurred before you ask it. 

>  #2

GamearooCodes, 2021-07-19:
sorry it asked me to report it when I clicked it auto made this for me

Bestduan, 2021-07-19:
Do you define arrays in your Verilog project, which can lead to a regex traceback trap

GamearooCodes, 2021-07-20:
i didnt really do anything this started to happen tho when I switched to windows 11

Bestduan, 2021-07-20:
So you just open an empty workspace and causes high cpu load?

GamearooCodes, 2021-07-20:
it seemed to fix itself idk what it was it may of been a bug with windows 11 but atm it seems to be working now

Bestduan, 2021-07-20:
OK， I will close the issue after that， if have other question， you can reopen it.

closed 2021-07-20 by Bestduan