Nitcloud/Digital-IDE #20: 使用`include命令引入宏定义时，使用iverilog仿真出错。

xlsjdjdk, 2021-08-25:
如题，我在RTL文件中引用了宏定义文件：
```
`include "defines.v"
```
该文件存放了形如
```
 `define     RstEnable       1'b1        //复位信号有效
 `define     RstDisable      1'b0        //复位信号无效
```
的宏定义。

此时在顶层文件右键点击simulate，会出现找不到宏定义文件的错误：
```
ERROR From iverilog : c:/Users/Desktop/mips_cpu/pc_reg.v:2: Include file defines.v not found
```
请问是否可以在iverilog编译命令中添加`-I includedir`选项以指定verilog中include指令的搜索路径呢？

Bestduan, 2021-08-25:
在下个版本会将该功能补充进去，不过近期工程在做第四次重构，可能不会很快上线。不知道您对此扩展的评价如何，是否实际解决您设计中的问题，是否还有其他的设计痛点呢？

xlsjdjdk, 2021-08-25:
感谢回复!我还在学习使用这个拓展,如果有其他问题,我会在issue反馈.目前观察到的一个小问题是,在使用拓展建立vivado工程时,如果原始文件没有按照工程格式组织,需要把文件放到生成的工程目录下并刷新,不知这个问题是否有更好的设计思路.

Bestduan, 2021-08-25:
emmm，正常设计思路不就是搭建工程结构，再添加设计文件吗，在vivado里也是这样的一个工程的啊，要不我再写一个自动识别HDL文件再移动到相应的文件夹下？主要这个看个人设计方式不是很好实现。不过唯一的想你原来有在vivado下创建的工程，可以将其转化为vscode下的标准工程，这个是比较好做的。

Bestduan, 2021-08-31:
iverilog不是可以自己去找include下的路径的吗？请参考 https://zhuanlan.zhihu.com/p/368089338

Bestduan, 2021-08-31:
> 如题，我在RTL文件中引用了宏定义文件：
>
> ```
> `include "defines.v"
> ```
>
> 该文件存放了形如
>
> ```
>  `define     RstEnable       1'b1        //复位信号有效
>  `define     RstDisable      1'b0        //复位信号无效
> ```
>
> 的宏定义。 [...]

我尝试了-I指令但是iverilog显示依旧找不到include下的文件，这很奇怪，请问你使用 -I 指令后是否有效？

Bestduan, 2021-09-05:
已解决，如果还有其他问题欢迎重开issue或者提出新的issue

closed 2021-09-05 by Bestduan