Nitcloud/Digital-IDE #22: 生成Vivado工程时读取配置出错问题

xlsjdjdk, 2021-08-29:
在设置好配置文件之后，仍按照默认模板生成vivado工程。
我在如下位置设置了配置文件：
`./property.json`,`./.vscode/property.json`,`prjInitParam.json`
文件内容：
```
{
	"FPGA_VERSION": "xilinx",
	"PRJ_NAME": {
		"FPGA": "mipscpu"
	},
	"SOC_MODE": {
		"soc": "none"
	},
	"enableShowlog": true,
	"Device": "none"
}
```
发现工程名称均为默认的template.
检查启动脚本`Run.tcl`，发现脚本读取了`.vscode\extensions\sterben.fpga-support-0.1.16\resources\tool`目录下的`CONFIG`文件。而此文件在创建配置文件时并未被修改。
```
set soc           none
set Device        none
set prj_name      template
set enableShowlog false
set fp [open $root_path/CONFIG r]
while { [gets $fp data] >= 0 } {
	if { [string equal -length 13 $data "PRJ_NAME.FPGA"] == 1 } {
		gets $fp prj_name
		if {$prj_name == "undefined"} {
			set prj_name template
		}
	}
...
```

Bestduan, 2021-08-29:
确实有问题，我刚刚查看了一下源码里没把更新函数加进去，这个问题比较严重，这几天找时间就修一下

Bestduan, 2021-09-05:
已解决

closed 2021-09-05 by Bestduan