Nitcloud/Digital-IDE #25: 提示语法错误的功能突然不行了

qgzln, 2021-09-08:
按了几个插件之后就突然不行了，现在将之前的无关插件都卸载了，也不行

Bestduan, 2021-09-08:
请提供更多信息，这些无法定位，比如使用的是哪个三方工具进行的语法检查，是否在setting中设置了其他内容，文件更改后是否保存之类。

qgzln, 2021-09-08:
使用的是vivado的语法检查，保存了。奇怪的地方在于出问题 的是我新建的文件和文件夹里，旧有的文件依然可以报错

qgzln, 2021-09-09:
今天我重新打开设置这里
[image]
先换回到default，然后再换回vivado，会出现
[image]
不知道是不是vscode出的问题，我重新安装了
[image]
语法检查一样没起作用

Bestduan, 2021-09-09:
我也定位到这了，也在找原因，会尽快修复。

Bestduan, 2021-09-09:
我已找到潜在bug点，但是我的语法检查一直都能用，无法问题复现，只能发布后(预计周末)等待您的反馈。
TerosHDL目前已更新到[2.0.0](https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest)
新加了不少三方支持但是原问题并没修复多少，重点是语法解析内核的不完善。

Bestduan, 2021-09-11:
已更新，请再次查看，是否已修复。

qgzln, 2021-09-13:
语法检查还是不行。之前有这么个现象，我发现你这个插件之前上个笔记本电脑一直用后来有你这个
[image]
之后就把他卸载了，当时语法检查就有问题了，后来我换了笔记本电脑，没有按这个插件是好的，我发现语法检查不行了之后就把这个插件有按上了

Bestduan, 2021-09-13:
我这边按你说的尝试了一下并没有出现插件不能用的情况，实在不行的话可以先用Verilog-HDL/SystemVerilog/Bluespec SystemVerilog，两个之间应该也不影响。之前我用的是Verilog-HDL/SystemVerilog/Bluespec SystemVerilog的检查方案，之后为了兼容更多的三方就换成了TerosHDL的方案，之后有时间我再检查吧，最近是没时间更新了。

qgzln, 2021-09-13:
ok

Bestduan, 2022-04-30:
新版本已经修复，请重新查看一下在你那是否有效

closed 2022-05-07 by Bestduan