Nitcloud/Digital-IDE #28: 使用iverilog在线仿真报错：找不到gtkwave

ZYZ9802, 2021-11-13:
[image]
使用iverilog在线仿真报错：找不到gtkwave；
但是我已经将gtkwave加入到系统环境变量,并且我在终端可以直接打开gtkwave:
[image]

Bestduan, 2021-11-13:
这个问题说明你在仿真文件里面没有说明仿真时间，也就是说没设置 `$finish` 语句，这时候iverilog里的vvp进程不会退出。所以说这个gtkwave相当于从vvp进程中打开，自然找不到。

解决办法是手动退出vvp进程，再手动输入gtkwave打开指令

另外就是在tb文件中添加 `$finish` 语句 设置仿真时间让vvp自动退出。

ZYZ9802, 2021-11-14:
已经解决了，感谢！还有个小问题：运行simulate时怎么可以让.vcd波形文件和.vvp一起生成在prj文件夹下呢？目前只有.vvp在prj下面

Bestduan, 2021-11-14:
vcd文件的创建是iverilog决定的，不是插件决定的，你可以尝试更改tb文件中的 `$dumpfile` 语句，具体申明创建到哪个路径，建议使用绝对路径，不然插件无法识别。

此外对于标准文件结构的分类，针对prj文件夹下的其实是可再生文件夹，就是说将整个prj文件删除也能在插件中还原出来，只要保证配置相同，这样做的目的一个是方便版本分发，另一个是设计时更加关注设计源文件（src文件夹下的内容）和输出文件（工程根目录下的内容），故不建议将输出文件vcd放置prj下。

ZYZ9802, 2021-11-15:
OK 感谢！

closed 2021-11-15 by ZYZ9802