Nitcloud/Digital-IDE #32: 例化功能不正常

hhucchenyixiao, 2021-12-21:
[image]

有的模块例化的时候，port列表总是会出现一些错误，有些port还重复出现，括号也莫名奇妙多，一些内部的reg也在这里显示。

Bestduan, 2021-12-21:
应该是 `input_key` 里面含有 `input` 这个敏感字段，被识别成端口了，改成 `key_i` 试试看有没有效果，不过这点我在写解释器的时候应该考虑到了，方便的话能复制出你的端口申明这段代码吗，有时间我做个测试。

hhucchenyixiao, 2021-12-22:
我改成key_in, data_in 后问题的确解决了。

你可以用这个代码测试
```verilog
module a (
    input  input_a,
    input  input_b,
    output output_c
);

    output_c = input_a & input_b;

endmodule
```

我试了下也可以触发这个bug
```verilog

wire  	_c;
wire  	output_c;
wire  	_c;

a u_a(
	//ports
	._c       		( _c       		),
	.input_b  		( input_b  		),
	._b       		( _b       		),
	.input_a  		( input_a  		),
	._a       		( _a       		),
	.input_b  		( input_b  		),
	._b       		( _b       		),
	.output_c 		( output_c 		),
	._c       		( _c       		)
);
```

hhucchenyixiao, 2021-12-22:
刚刚我发现例化还有个错误
···verilog
output reg  [`CLOG2(MEM_DEPTH) - 1:0] mem_address,
```
被例化成了
```verilog
wire [`CLOG2[MEM_DEPTH]-1:0] mem_address          ;
```
里面应该是( )但变成了[ ]

Bestduan, 2021-12-22:
这个写法确实没考虑到，下个版本修复。

closed 2021-12-27 by hhucchenyixiao