Nitcloud/Digital-IDE #33: crashes my extension host

devriesewouter89, 2022-01-14:
I used vscode bisect to identify the error. Each time at startup of vscode, it kills my extension host.

Bestduan, 2022-01-14:
Could you describe the problem you met in detail?  I can't locate the problem according to your description.  Can you take screenshots or paste error logs.

devriesewouter89, 2022-01-14:
Hi @Bestduan , sorry, I did forget the attach

[image]

I get the error message when launching vscode with plugin enabled. If I disable the plugin my extension host doesn't crash. 
Yet, if I disable the plugin, launch vscode and THEN launch this plugin, all goes fine.

Bestduan, 2022-01-14:
From your screenshots I think this may not be my problem. First, I did not find any errors related to my extension, and second, according to the error information, it is the extension called `regex workbench` that throw the error.  You can try to disable it. If you still have the same problem after this, please inform me.

Bestduan, 2022-01-17:
May I ask whether the previous reply has solved your problem?

devriesewouter89, 2022-01-17:
Hi

you were correct, sorry for the inconvenience, strange that VSCode pointed
your implementation as the wrong one...

Bestduan, 2022-01-17:
I am glad that you have solved the problem and the new version has been released. Please let me know if you have any other problems. Thank you for your support.

closed 2022-01-17 by Bestduan