Nitcloud/Digital-IDE #49: 希望支持括号辅助

joson1, 2022-07-14:
效果类似于：

[image]

光标停在begin处会自动框起来end，反过来也是可以的

Bestduan, 2022-07-14:
[image]
插件是支持框的，但从end处括起来应该是vscode本身不支持

joson1, 2022-07-24:
我提的插件是这款 https://marketplace.visualstudio.com/items?itemName=mshr-h.VerilogHDL
让vscode将begin end 当 { } 来处理，还支持colourful brackets，就是不通层级的括号 颜色不同
[image]
[image]

joson1, 2022-07-24:
[image]
这个插件目前已经集成在最新版的VS code中了

Bestduan, 2022-07-25:
好的，我试一下

Bestduan, 2022-07-28:
新版本0.2.5已经修复该问题，请check out一下。

joson1, 2022-08-04:
太赞了！

closed 2022-08-04 by joson1