Nitcloud/Digital-IDE #50: Tool commands not found.

Tsegorah1, 2022-07-28:
I've installed Digital IDE plugin in VSC, windows 10, and tried to generate project.
[image]

Bestduan, 2022-07-28:
This Error means this extension did not launch. Click `help` -> `Toggle Developer Tools` , and check out the `Cnsole`, if there are some errors.

Tsegorah1, 2022-07-29:
This is what it shows:
 INFO UNRESPONSIVE extension host: starting to profile NOW
log.ts:307  WARN UNRESPONSIVE extension host: 'sterben.fpga-support' took 93% of 5452.182ms, saved PROFILE here: 'file:///c%3A/Users/PC/AppData/Local/Temp/exthost-1bc96f.cpuprofile' (2) [{…}, {…}]
There is no such errors with other extensions like VHDL LS.

Tsegorah1, 2022-07-29:
I've solved this problem after setting PRJ file limit number from 5000 to 200.
Can I ask another question here? What should I write in property.json to make the extension see my files? I tried:
"PRJ_STRUCTURE": "customer",
	"PRJ_Path": "${workspace}",
	"HardwareSrc": "${workspace}"
but it gives an error:
Activating extension 'sterben.fpga-support' failed: Cannot read properties of undefined (reading 'replace').

Bestduan, 2022-07-29:
You have missed some properties that were not set, the full example is as follows:
```json
{
    "TOOL_CHAIN": "xilinx",
    "PRJ_NAME": {
        "FPGA": "temp"
    },

    "PRJ_STRUCTURE": "customer",

    "PRJ_Path": "D:/Project",
    "HardwareSrc": "${workspace}",
    "HardwareSim" : "${workspace}",
    "HardwareData" : "${workspace}",
    "SoftwareSrc" : "${workspace}",
    "SoftwareData" : "${workspace}",

    "SOC_MODE": {
        "soc": "none"
    },
    "enableShowlog": false,
    "Device": "xc7a35tcsg324-1"
}
```
`"PRJ_Path"` does not support `"${workspace}"`, it needs an absolute path.
Finally I don't recommend setting the project path and the source path to be the same, this may cause the plugin to get stuck.

closed 2022-07-29 by Tsegorah1