Nitcloud/Digital-IDE #53: 关闭GUI后报错，无法识别add_file命令

EriccWuu, 2022-09-16:
通过插件创建vivado工程后打开GUI，给工程加了个IP核，关掉工程后抱了一堆错，都是一个类型。报错信息如下：
add_file : 无法将“add_file”项识别为 cmdlet、函数、脚本文件或可运行程序的名称。请检查名称的拼写，如果包括路径，请确保路径正确，然后再试一次。
是我没配置好什么嘛？识别不出这个命令

EriccWuu, 2022-09-16:
再补充一点东西。再通过插件打开vivado GUI就看不到之前加的IP核了，但是不通过插件直接打开工程就可以看到。

Bestduan, 2022-09-17:
一开始报错是属于正常情况，在生成IP之后会生成一些HDL文件，被插件检测到了，由于vivado的阻塞没当时执行，vivado关闭之后就会生成add_file的命令，但vivado已经关闭不在tcl的环境下所以会有失败提示，不用管就行了。另外再次打开没加入IP的话是不是你没有使用标准文件结构？

EriccWuu, 2022-09-17:
报错的原因和我也想的差不多。至于再次打开没有IP的话，我用的是标准文件结构，而且是默认的没改，但我加IP的时候，IP的存放位置是默认的，没有按照插件的标准文件结构存在user/ip里，之后不论通过插件打开，还是直接打开工程都看不到加的IP。收到你的回复后我又试着把IP放在了user/ip里，然后不论通过什么方式打开工程都可以看到IP了。因为我看知乎教程里说存在vivado管理的ip会被插件搬移到user/ip下，我就以为加IP得时候位置默认就好了。

DreamLand7707, 2023-07-29:
> 报错的原因和我也想的差不多。至于再次打开没有IP的话，我用的是标准文件结构，而且是默认的没改，但我加IP的时候，IP的存放位置是默认的，没有按照插件的标准文件结构存在user/ip里，之后不论通过插件打开，还是直接打开工程都看不到加的IP。收到你的回复后我又试着把IP放在了user/ip里，然后不论通过什么方式打开工程都可以看到IP了。因为我看知乎教程里说存在vivado管理的ip会被插件搬移到user/ip下，我就以为加IP得时候位置默认就好了。
> […](#)

我同样遇到了这个问题，请问您复制IP复制的是哪个文件？.xdi还是某个文件夹？
我目前为了能够保证IP核识别的正确性不得不自己编写tcl文件来控制某些文件的加入和工程的创建，而是用插件的目的仅仅是为了保证代码提示和补全。而不是launch gui synth的功能。

EriccWuu, 2023-07-29:
当时就是把ip删了重新创建，位置按照插件要求的文件结构。不过我现在也基本上就用个代码补全和提示，以及快速例化了。

发件人: DrLmow
发送时间: 2023年7月29日 17:20
收件人: Nitcloud/Digital-IDE
抄送: Eric Wu; Author
主题: Re: [Nitcloud/Digital-IDE] 关闭GUI后报错，无法识别add_file命令 (Issue #53)

报错的原因和我也想的差不多。至于再次打开没有IP的话，我用的是标准文件结构，而且是默认的没改，但我加IP的时候，IP的存放位置是默认的，没有按照插件的标准文件结构存在user/ip里，之后不论通过插件打开，还是直接打开工程都看不到加的IP。收到你的回复后我又试着把IP放在了user/ip里，然后不论通过什么方式打开工程都可以看到IP了。因为我看知乎教程里说存在vivado管理的ip会被插件搬移到user/ip下，我就以为加IP得时候位置默认就好了。
…
------------------ 原始邮件 ------------------ 发件人: @.>; 发送时间: 2022年9月17日(星期六) 下午2:57 收件人: @.>; 抄送: @.>; @.>; 主题: Re: [Bestduan/Digital-IDE] 关闭GUI后报错，无法识别add_file命令 (Issue #53) 一开始报错是属于正常情况，在生成IP之后会生成一些HDL文件，被插件检测到了，由于vivado的阻塞没当时执行，vivado关闭之后就会生成add_file的命令，但vivado已经关闭不在tcl的环境下所以会有失败提示，不用管就行了。另外再次打开没加入IP的话是不是你没有使用标准文件结构？
我同样遇到了这个问题，请问您复制IP复制的是哪个文件？.xdi还是某个文件夹？
我目前为了能够保证IP核识别的正确性不得不自己编写tcl文件来控制某些文件的加入和工程的创建，而是用插件的目的仅仅是为了保证代码提示和补全。而不是launch gui synth的功能。