Nitcloud/Digital-IDE #57: "show netlist" option cannot show correctly when other module instantiation in top module

Collect-diamond, 2023-04-08:
I defined 2 verilog modules, one for DFF and one for EXAMP, I instantiated the DFF module twice in the EXAMP module, but when I use the "show netlist" option to view the schematic, it shows the schematic of the DFF module, while when I change the name of the EXAMP module to A (the initials of the module are sorted before D), the schematic is correct. is it a bug?
[image]
[image]

Nitcloud, 2023-04-10:
The issue has been logged, can you please provide sample code for testing?

Collect-diamond, 2023-04-11:
D_FF.v
```
module D_FF(clk,d,q);
    output reg q;
    input clk,d;
    always @ (posedge clk)
        q<=d;
endmodule
```

TEST.v (**module A or E**)
```c
module E (clk,d,a,q);
    output  q;
    input clk,d,a;

    wire d1;
    wire q1;

    D_FF dff1(.clk(clk),.d(d1),.q(q1));
    D_FF dff2(.clk(q1),.d(d),.q(q));

    or (d1,a,q);

endmodule
```